CIDARLAB/3DuF #41: Avoid using delete when removing features

Nesciosquid, 2015-12-15:
The Layer class has method ```removeFeatureByID(featureID)``` which uses the ```delete``` operation to remove a feature from its ```features``` array. 

Delete is slow! Find a better way to store features to improve performance.

tjsangster, 2020-12-17:
I believe [Map](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Map) would be better suited for the 'features' property of this class than [Object](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Object). If this property was updated to Map we could remove this delete operation as well as the 'featureCount' property.

rkrishnasanka, 2020-12-18:
@tjsangster yeah that would be a possibility, we might need to switch up the drawing architecture at a later point. Right now we heavily rely on the features object for hit-testing and a bunch and keeping track of what's being rendered. But keep the ideas flowing, it'll be good to revisit them.