CIDARLAB/3DuF #67: 3DuF does not flip control layer

mzw23, 2017-04-22:
The SVG of the control layer must be flipped for correct milling. Currently 3DuF does not do this.

### Current 3DuF Control Output Example
[image]

### Correct 3DuF Control Output Example
[image]

rkrishnasanka, 2017-11-08:
I'd like to bring this up again as an issue. Earlier versions of 3DuF used to flip the chip horizontally. That needs to be reimplemented again.

rkrishnasanka, 2018-08-31:
Implemented in the CNC outputs as of commit - 2545e9e5107388d6f9d0fec27bfa0b405c68d20f

closed 2018-08-31 by rkrishnasanka