CIDARLAB/3DuF #83: Rotating primitives

dinithi-s, 2017-06-15:
The ability to set  primitive on the grid and then rotate it would be extremely helpful - especially when creating a fan branch or circular designs.

rkrishnasanka, 2018-05-31:
We have the 360-degree rotation now for Gradient Generator. We need to figure out how we can incorporate this into all the other components. We also need to figure out a strategy for accommodating the older files. Perhaps a script would be sufficient for converting all the other JSON features.

rkrishnasanka, 2019-04-27:
So we are slowly transitioning all the components to have 360-degree rotations and including code to modify the design files automatically without the need of an external script.

closed 2021-06-18 by rkrishnasanka