CIDARLAB/3DuF #86: Mixer does not mill properly

sarahnem, 2017-06-23:
When trying to mill out this design:
[image]
The mixers did not mill out correctly:
[image]
When milling, the mill reads each part of the mixer as a separate channel, not as one entire feature, so it does not cut it out correctly. Using an exacto knife to cut the excess polycarb blocking the channels leads to the chip becoming mangle and not uniform, which is an issue.
Multiple people have encountered this issue with endmills of varying sizes. 
Ideally, the mill would follow the path of the mixer when milling instead of cutting individual channels and then (failing to properly) connect them. I'm not sure how to make that happen, but if it could that would be great.

sivwizinbiznilva, 2017-06-23:
This can be mitigated by performing an SVG union, as it seems 3DuF export each vector as an individual entity.

sarahnem, 2017-06-23:
How do you perform an SVG union?

sivwizinbiznilva, 2017-06-23:
https://imagesbyheatherm.wordpress.com/2010/01/15/inkscape-path-operations-explained-part-1/

Union or stroke-to-path should do the trick! Inkscape is free.

JLippai, 2017-07-14:
There is an experimental fix for this that currently hurts some of the live-update functionality of the sliders. Will test it more and iron out kinks.

JLippai, 2017-08-25:
The new live "Better Mixer" primitive resolves this issue, and will replace the old Mixer as the only square Mixer primitive in the next couple of days

closed 2017-08-25 by JLippai