CIDARLAB/3DuF #88: Inputting JSON Issue

sarahnem, 2017-06-29:
Importing JSON files into 3Duf no longer works properly:
[image]
[image]
The issues listed vary each time; here's the JSON that I was attempting to import.
`{"name":"treelayer","layers":[{"features":{"57e0850c-3033-4ec5-b275-bab5c58e252c":{"set":"Basic","name":"vc","id":"57e0850c-3033-4ec5-b275-bab5c58e252c","type":"Valve3D","params":{"orientation":"V","valveRadius":50,"gap":25,"width":100,"length":100,"position":[1175,2484]}},"7f7abfcd-eed4-4976-b7ba-e883f46923a4":{"set":"Basic","name":"ve","id":"7f7abfcd-eed4-4976-b7ba-e883f46923a4","type":"Valve3D","params":{"orientation":"V","valveRadius":50,"gap":25,"width":100,"length":100,"position":[3175,2486]}},"268f4b0e-bfdc-48ee-98db-eda0b1825b4a":{"set":"Basic","name":"controlport","id":"268f4b0e-bfdc-48ee-98db-eda0b1825b4a","type":"Port","params":{"portRadius":50,"position":[5221,125]}},"fe206c67-4cb0-4e6b-9c59-18fb64c0053a":{"set":"Basic","name":"va","id":"fe206c67-4cb0-4e6b-9c59-18fb64c0053a","type":"Valve3D","params":{"orientation":"V","valveRadius":50,"gap":25,"width":100,"length":100,"position":[1675,1540]}},"fedef4be-cfd2-464d-81b6-94d0c09cb6b1":{"set":"Basic","name":"vb","id":"fedef4be-cfd2-464d-81b6-94d0c09cb6b1","type":"Valve3D","params":{"orientation":"V","valveRadius":50,"gap":25,"width":100,"length":100,"position":[175,2487]}},"e39a836f-f829-45e4-b403-b596985a9277":{"set":"Basic","name":"vd","id":"e39a836f-f829-45e4-b403-b596985a9277","type":"Valve3D","params":{"orientation":"V","valveRadius":50,"gap":25,"width":100,"length":100,"position":[2175,2484]}}},"color":"red","name":"control","type":"control","params":{"flip":true,"z_offset":1200}},{"features":{"2d945b15-f176-41a9-a571-0461256ba1b3":{"set":"Basic","name":"c1a","id":"2d945b15-f176-41a9-a571-0461256ba1b3","type":"Channel","params":{"start":[1725,1339],"end":[1725,1540],"channelWidth":12}},"6c274a3e-fcd6-4c52-a44c-b84631e51f69":{"set":"Basic","name":"vb","id":"6c274a3e-fcd6-4c52-a44c-b84631e51f69","type":"Valve3D","params":{"orientation":"V","valveRadius":50,"gap":25,"width":100,"length":100,"position":[175,2487]}},"123edc15-8831-4e9d-b276-50856b1006bd":{"set":"Basic","name":"flowtree","id":"123edc15-8831-4e9d-b276-50856b1006bd","type":"Tree","params":{"flowChannelWidth":12,"orientation":"H","spacing":1000,"leafs":4,"width":3200,"length":444,"position":[125,1840],"direction":"OUT"}},"0628c7a5-e9c9-4d41-babe-da2766c3273a":{"set":"Basic","name":"c1c","id":"0628c7a5-e9c9-4d41-babe-da2766c3273a","type":"Channel","params":{"start":[1225,2788],"end":[1225,2584],"channelWidth":12}},"f92b6b9f-e237-4872-affa-74b91f72e835":{"set":"Basic","name":"a","id":"f92b6b9f-e237-4872-affa-74b91f72e835","type":"Port","params":{"portRadius":50,"position":[1675,1239]}},"c3314074-188c-41f0-a259-620992e11cbd":{"set":"Basic","name":"b","id":"c3314074-188c-41f0-a259-620992e11cbd","type":"Port","params":{"portRadius":50,"position":[175,2787]}},"411f85c7-702d-4643-8644-1f6207e29fd2":{"set":"Basic","name":"ve","id":"411f85c7-702d-4643-8644-1f6207e29fd2","type":"Valve3D","params":{"orientation":"V","valveRadius":50,"gap":25,"width":100,"length":100,"position":[3175,2486]}},"e55de4b4-28a5-49b2-8353-c7979716213f":{"set":"Basic","name":"c2a","id":"e55de4b4-28a5-49b2-8353-c7979716213f","type":"Channel","params":{"start":[1725,1640],"end":[1725,1840],"channelWidth":12}},"8572b807-37ac-431b-8877-b9bfd49fbda6":{"set":"Basic","name":"c1d","id":"8572b807-37ac-431b-8877-b9bfd49fbda6","type":"Channel","params":{"start":[2225,2785],"end":[2225,2584],"channelWidth":12}},"4066fd37-bfbe-456a-86e4-3a49b294169c":{"set":"Basic","name":"c2d","id":"4066fd37-bfbe-456a-86e4-3a49b294169c","type":"Channel","params":{"start":[2225,2484],"end":[2225,2284],"channelWidth":12}},"7b1982ef-3198-4723-96ef-6906db107e83":{"set":"Basic","name":"va","id":"7b1982ef-3198-4723-96ef-6906db107e83","type":"Valve3D","params":{"orientation":"V","valveRadius":50,"gap":25,"width":100,"length":100,"position":[1675,1540]}},"1e366ebd-2bad-41d4-9ad3-f796b87de330":{"set":"Basic","name":"c1b","id":"1e366ebd-2bad-41d4-9ad3-f796b87de330","type":"Channel","params":{"start":[225,2787],"end":[225,2587],"channelWidth":12}},"d396adb4-732a-4f59-bf5a-6e96d8e36136":{"set":"Basic","name":"vd","id":"d396adb4-732a-4f59-bf5a-6e96d8e36136","type":"Valve3D","params":{"orientation":"V","valveRadius":50,"gap":25,"width":100,"length":100,"position":[2175,2484]}},"94992964-2151-455d-85b8-83c7bb3202b8":{"set":"Basic","name":"c1e","id":"94992964-2151-455d-85b8-83c7bb3202b8","type":"Channel","params":{"start":[3225,2786],"end":[3225,2586],"channelWidth":12}},"73614f15-5501-4e89-acf8-683137f215dd":{"set":"Basic","name":"vc","id":"73614f15-5501-4e89-acf8-683137f215dd","type":"Valve3D","params":{"orientation":"V","valveRadius":50,"gap":25,"width":100,"length":100,"position":[1175,2484]}},"0e86e846-f916-4c2c-8aa9-7291b4bca952":{"set":"Basic","name":"c2e","id":"0e86e846-f916-4c2c-8aa9-7291b4bca952","type":"Channel","params":{"start":[3225,2486],"end":[3225,2284],"channelWidth":12}},"80548d7e-1696-4ce3-af86-4ca26fd1d19d":{"set":"Basic","name":"d","id":"80548d7e-1696-4ce3-af86-4ca26fd1d19d","type":"Port","params":{"portRadius":50,"position":[2175,2785]}},"bb25ce9f-fb06-4f15-addb-90f39473c3b4":{"set":"Basic","name":"c2b","id":"bb25ce9f-fb06-4f15-addb-90f39473c3b4","type":"Channel","params":{"start":[225,2487],"end":[225,2284],"channelWidth":12}},"c9b51fba-c5f6-49a4-a64e-bde54c4ead92":{"set":"Basic","name":"c2c","id":"c9b51fba-c5f6-49a4-a64e-bde54c4ead92","type":"Channel","params":{"start":[1225,2484],"end":[1225,2284],"channelWidth":12}},"cffe2571-24aa-4f1d-a983-99c0f7b3c5a4":{"set":"Basic","name":"c","id":"cffe2571-24aa-4f1d-a983-99c0f7b3c5a4","type":"Port","params":{"portRadius":50,"position":[1175,2788]}},"0a3e3ebf-cb16-47a5-982e-16423b87b091":{"set":"Basic","name":"e","id":"0a3e3ebf-cb16-47a5-982e-16423b87b091","type":"Port","params":{"portRadius":50,"position":[3175,2786]}}},"color":"indigo","name":"flow","type":"flow","params":{"flip":false,"z_offset":0}}],"params":{"lambda":40,"width":215840,"total_wire_length":80440.0,"utilization":-12.912279367114918,"avg_wire_length":8044.0,"total_chip_area":-1.88446976E8,"height":118520,"used_area":2.43328E9}}`

rkrishnasanka, 2018-06-18:
We do not use the parameters used in this version of TREE anymore.  I am converting this issue into a feature request and am going to start referencing all issues with incorrect JSON import to here.

rkrishnasanka, 2021-06-18:
Will be handled in #284

closed 2021-06-18 by rkrishnasanka