CIDARLAB/3DuF #118: Valve3D does not delete properly

zheins, 2018-04-17:
[image]

@JLippai

rkrishnasanka, 2018-04-18:
@JLippai this is happening because by default, its only selecting the active layer components. We need to enable multilayer selection for single right click selection.

rkrishnasanka, 2018-05-12:
I have fixed the selection system to include the multiple features on the canvas via 'component' abstraction. I haven't however tested #121 with regards to this. I will look at it next after @JLippai has finished the selection tool. I already made changes to it and I don't want to make all the changes right now. However the deletion works fine now !!!

closed 2018-05-12 by rkrishnasanka