CIDARLAB/3DuF #124: Transposer Issues

maniwu, 2018-06-04:
- Missing connections between valves on control layer

maniwu, 2018-06-05:
The control layer given in 3DuF does not have channels connecting the valves to a port for pressure control. 
[image]
[image]

rkrishnasanka, 2018-06-13:
@maniwu This has been partially fixed (the inside channels are done). It would be good if you all parameterized component so that this does not remain to be a problem.

closed 2021-04-03 by rkrishnasanka