CIDARLAB/3DuF #127: Flipping flow or control layer for svg files

maniwu, 2018-06-21:
When creating chip designs, either the flow or control layer should be flipped since, during assembly, either of the layers will be flipped in order to have the milled faces align. 

For example, a milled chip below has valves (in the center of the design) that do not align on the flow and control layers despite appearing aligned on the json and svg files.

[image]

[image]

rkrishnasanka, 2018-06-22:
@maniwu did you not flip the control layer before fabricating? I thought the rest of the group does this, Ideally 3DuF needs to do this but I assumed that it was part of standard procedure.

rkrishnasanka, 2018-06-22:
Also, the issue is already in the system - refer to issue #67 . Please try to extend existing issues.

closed 2018-06-22 by rkrishnasanka