CIDARLAB/3DuF #139: Can't use move tool

jblugagne, 2018-10-16:
When I select an element with the select tool, and click again, the "element menu" pops up. However when I click on the "move" icon, nothing happens. I tried long presses, dragging around etc...

rkrishnasanka, 2018-10-17:
I've addressed this in commit 86a8ea646b6f45082f63185c77142ef950f43e69. This is live right now so let me know if this is working correctly.

jblugagne, 2018-10-17:
Actually this isn't really fixed. Right now, if I start anew and start placing connections or channels, the move tool still does nothing when I click it.

However, if I start placing other objects, chambers or mixers for example, I can select them and use the move tool on **those**. And then, once I've done that, the move tool starts working on the connections and the channels.

It does not work if I place connections/channels, then chambers or whatever (without trying to move them) and then try directly to move the connections/channels. I have to try to move the chambers first, and then the move tool starts working on connections.

rkrishnasanka, 2018-10-17:
So, the live version of it is working -  to this extent.

[image]

I haven't yet implemented the move tool for the connections, I need to figure out how exactly I want to enable it for that.

jblugagne, 2018-10-17:
Great!
A small comment though: Once you start moving the element "live", the X and Y values in the pop-up menu jump from something like 20mm to tens of thousands. You can see that in the gif you just posted.

rkrishnasanka, 2018-10-18:
Whoops, I forgot the unit conversion! I'll fix that