CIDARLAB/3DuF #148: Building dev branch has unmet dependencies

jsreid13, 2018-12-04:
I'm building the dev branch by running npm install after installing node and npm and I'm running into some unmet dependencies that I'm not sure if I'm solving correctly:

- I installed node v11, not sure if that is the version you are using
- This required Babel which I installed using ```npm install --save-dev @babel/core @babel/preset-env``` after the following error:
```
Error: Cannot find module '@babel/core'
 babelify@10 requires Babel 7.x (the package '@babel/core'). If you'd like to use Babel 6.x ('babel-core'), you should install 'babelify@8'.
```
- This then caused 3 module not found errors due to the first letter of the file being capitalized (view/render3D/ThreeDeviceRenderer, ./PanAndZoom and ./render2D/GridRenderer which after copying these files in the same directory with the first letter capitalized resolved the issue, not sure if this is an issue with a setting within Babel
- After solving these issues it did install, however trying to run the tests resulted in another unmet dependency of mocha, after npm installing that it raised an error of not being able to find the module babel/register

And this is where I'm stuck now, do you have any suggestions?

rkrishnasanka, 2018-12-05:
Hello @jsreid13 

Installation Errors
----
Looks like you need to have admin privileges to be able to the installation in 1 shot. This command should work without any issues: `sudo npm install`.

This would not require you to do any modifications to any of the files. I'm not yet certain why you need admin privs but it seems like the babel installation doesn't go through fully if you don't do this. Test in on a fresh clone of 3DuF. I just tested this on Mac but haven't been able to test this on Windows. 

Tests
---
Unfortunately, our tests haven't been updated in almost two years. With constant changes and upgrades to internal data structures, it was hard to keep up both the development test case generation.

rkrishnasanka, 2018-12-13:
@jsreid13 Did the instructions help?

jsreid13, 2018-12-14:
Sorry for the delay! I've been busy with finals, I decided to run this inside of a Node Docker container so that the environment is consistent. Even when running the command within Docker (which automatically provided sudo privileges) it still ran into the same issues, needing Babel and the files renamed. Here's a copy of the Dockerfile I created, if you're familiar with how Docker works then feel free to play around with it to let me know what to do to simplify the installation process. 

This is the minimum required to run ```npm install```:
```
FROM node:11.4.0-stretch
RUN git clone https://github.com/CIDARLAB/3DuF.git /home/3DuF
WORKDIR /home/3DuF
RUN npm install
```
However calling ```npm run compile``` requires modifications, these are some commands that I attempted but it still does not compile, saying that it cannot find the file ./src/app/utils/SimpleQueue, do you have any suggestions?
```
FROM node:11.4.0-stretch
RUN git clone https://github.com/CIDARLAB/3DuF.git /home/3DuF
WORKDIR /home/3DuF
RUN npm install --save-dev @babel/core @babel/preset-env
RUN cp ./src/app/view/render3D/threeDeviceRenderer.js ./src/app/view/render3D/ThreeDeviceRenderer.js
RUN cp ./src/app/view/panAndZoom.js ./src/app/view/PanAndZoom.js
RUN cp ./src/app/view/render2D/gridRenderer.js ./src/app/view/render2D/GridRenderer.js
RUN npm install
RUN cpm run compile
```

rkrishnasanka, 2018-12-14:
Hmmm, it seems like its actually having trouble compiling in Linux. Let me standardize some of the mismatched imports and try it again.

rkrishnasanka, 2018-12-16:
@jsreid13 So I managed to verify that the build system works in windows too. It seems to have some trouble with linux only for some weird reason. I'll keep you posted.

rkrishnasanka, 2018-12-17:
So I realized that the entire issue came on because of the wrong case usage in the file names used in the imports. Using the exact case of the filename in the imports fixed everything. So the project is building as of commit 0eebf438faa4b652b30a41b0fac2d4160615fb41 now available in master too.

closed 2018-12-17 by rkrishnasanka