CIDARLAB/3DuF #150: Unable to use move tool

dpmcintyre7, 2019-01-16:
Some components are unable to move after a while.

From chrome devtools:

Both right and left tool is the same
3DuFapp.js:90768 Uncaught TypeError: Cannot read property 'getPosition' of null
    at MoveTool.activate (3DuFapp.js:90768)
    at RightClickMenu.__activateMove (3DuFapp.js:94657)
    at HTMLDivElement.<anonymous> (3DuFapp.js:94552)
533DuFapp.js:90792 Uncaught TypeError: Cannot read property 'updateComponetPosition' of null
    at MoveTool.processUIPosition (3DuFapp.js:90792)
    at MoveTool.__updatePosition (3DuFapp.js:90804)
    at MoveTool.dragHandler (3DuFapp.js:90842)
    at MoveTool._this.move (3DuFapp.js:90746)
    at HTMLCanvasElement.<anonymous> (3DuFapp.js:84185)

rkrishnasanka, 2019-01-17:
@dpmcintyre7 please put of a screenshot of the canvas also so I know which component is throwing this error.

dpmcintyre7, 2019-01-17:
[image]

I just caused the error again, by selecting multiple different components, moving them, and pressing cancel repeatedly. After awhile, I can no longer press cancel and all components in the design cannot be moved unless I save, refresh 3DuF, and upload the design again.

rkrishnasanka, 2019-03-09:
This was fixed in v1.1. The issue was that some selection internally was getting messed up causing all the problems.

closed 2019-03-09 by rkrishnasanka