CIDARLAB/3DuF #162: connection's parameter manipulation

h12390, 2020-02-28:
updateParameter(key, value) in connection.js not work as expected. Need to fix it soon. We also need to double-check the rest of the parameter-related code.

rkrishnasanka, 2020-02-28:
Ok looks like we found the differences in what the `getParams()` returns the `__params.parameters` and `__params.parameters`

In `features.js`:

```
    getParams(){
        return this.__params.parameters;
    }
```

In `connection.js`:

```
    getParams(){
        return this.__params;
    }
```

In `component.js`:

```
    getParams(){
        return this.__params;
    }
```

In `textfeature.js`:
```
    getParams(){
        return this.__params.parameters;
    }
```

rkrishnasanka, 2020-02-28:
I will change the function name here

rkrishnasanka, 2021-06-21:
@murrayLF this was one of the problems with how the params API is being used

rkrishnasanka, 2021-08-19:
This has been rectified with the typescript port, the return type hints should make things more clear

closed 2021-08-19 by rkrishnasanka