CIDARLAB/3DuF #179: SVG file output is not generating correctly

jainadibu, 2020-04-18:
Developed the device as shown here: 
[image]

However, when exporting as SVG, the files appear to be empty/not generated correctly. The laser cutting and CNC files load accurately, as does the JSON, however the vector art export is not working.
[lasercutterV1.zip](https://github.com/CIDARLAB/3DuF/files/4497536/lasercutterV1.zip)
[device_layers.zip](https://github.com/CIDARLAB/3DuF/files/4497537/device_layers.zip)
[cncV1.zip](https://github.com/CIDARLAB/3DuF/files/4497538/cncV1.zip)

JSON output:
`{"name":"My Other Device","params":{"width":135000,"length":85000},"components":[{"id":"1eb2ed80-819e-11ea-9e5d-9b01e8e92c44","name":"CurvedMixer_1","entity":"CURVED MIXER","params":{"position":[115000,12000],"bendSpacing":6000,"numberOfBends":6,"channelWidth":2000,"bendLength":12000,"orientation":"H","height":1170},"xspan":97999.99999999999,"yspan":22000,"ports":[]},{"id":"18c4b830-819f-11ea-9e5d-9b01e8e92c44","name":"Port_2_8_1","entity":"PORT","params":{"position":[100000.00000000004,45000],"portRadius":1000,"height":1100},"xspan":2000,"yspan":2000,"ports":[{"x":0,"y":0,"layer":"FLOW","label":"1"}]},{"id":"18c5a290-819f-11ea-9e5d-9b01e8e92c44","name":"Port_2_8_2","entity":"PORT","params":{"position":[100000.00000000004,47500],"portRadius":1000,"height":1100},"xspan":2000,"yspan":2000,"ports":[{"x":0,"y":0,"layer":"FLOW","label":"1"}]},{"id":"18c665e3-819f-11ea-9e5d-9b01e8e92c44","name":"Port_2_8_3","entity":"PORT","params":{"position":[100000.00000000004,50000],"portRadius":1000,"height":1100},"xspan":2000,"yspan":2000,"ports":[{"x":0,"y":0,"layer":"FLOW","label":"1"}]},{"id":"18c77753-819f-11ea-9e5d-9b01e8e92c44","name":"Port_2_8_4","entity":"PORT","params":{"position":[100000.00000000004,52500],"portRadius":1000,"height":1100},"xspan":2000,"yspan":2000,"ports":[{"x":0,"y":0,"layer":"FLOW","label":"1"}]},{"id":"18c861b0-819f-11ea-9e5d-9b01e8e92c44","name":"Port_2_8_5","entity":"PORT","params":{"position":[100000.00000000004,55000],"portRadius":1000,"height":1100},"xspan":2000,"yspan":2000,"ports":[{"x":0,"y":0,"layer":"FLOW","label":"1"}]},{"id":"18c94c11-819f-11ea-9e5d-9b01e8e92c44","name":"Port_2_8_6","entity":"PORT","params":{"position":[100000.00000000004,57500],"portRadius":1000,"height":1100},"xspan":2000,"yspan":2000,"ports":[{"x":0,"y":0,"layer":"FLOW","label":"1"}]},{"id":"18ca3673-819f-11ea-9e5d-9b01e8e92c44","name":"Port_2_8_7","entity":"PORT","params":{"position":[100000.00000000004,60000],"portRadius":1000,"height":1100},"xspan":2000,"yspan":2000,"ports":[{"x":0,"y":0,"layer":"FLOW","label":"1"}]},{"id":"18cb20d3-819f-11ea-9e5d-9b01e8e92c44","name":"Port_2_8_8","entity":"PORT","params":{"position":[100000.00000000004,62500],"portRadius":1000,"height":1100},"xspan":2000,"yspan":2000,"ports":[{"x":0,"y":0,"layer":"FLOW","label":"1"}]},{"id":"a9e81b20-81a1-11ea-9e5d-9b01e8e92c44","name":"Port_3","entity":"PORT","params":{"position":[112000,82000],"portRadius":1500,"height":1100},"xspan":3000,"yspan":3000,"ports":[{"x":0,"y":0,"layer":"FLOW","label":"1"}]},{"id":"b1c27390-81a1-11ea-9e5d-9b01e8e92c44","name":"Port_4","entity":"PORT","params":{"position":[121000,82000],"portRadius":1500,"height":1100},"xspan":3000,"yspan":3000,"ports":[{"x":0,"y":0,"layer":"FLOW","label":"1"}]},{"id":"97182b10-81a2-11ea-9e5d-9b01e8e92c44","name":"Chamber_1","entity":"REACTION CHAMBER","params":{"position":[8000,29000],"width":9000,"length":25500,"height":250,"cornerRadius":200},"xspan":9000,"yspan":25500,"ports":[{"x":0,"y":-12750,"layer":"FLOW","label":"1"},{"x":4500,"y":0,"layer":"FLOW","label":"2"},{"x":0,"y":12750,"layer":"FLOW","label":"3"},{"x":-4500,"y":0,"layer":"FLOW","label":"4"}]},{"id":"728e3ce0-81a6-11ea-9e5d-9b01e8e92c44","name":"Port_5","entity":"PORT","params":{"position":[70000,45000],"portRadius":1000,"height":1100},"xspan":2000,"yspan":2000,"ports":[{"x":0,"y":0,"layer":"FLOW","label":"1"}]},{"id":"753abf40-81a6-11ea-9e5d-9b01e8e92c44","name":"Port_6","entity":"PORT","params":{"position":[70000,48000],"portRadius":1000,"height":1100},"xspan":2000,"yspan":2000,"ports":[{"x":0,"y":0,"layer":"FLOW","label":"1"}]},{"id":"78e8ac10-81a6-11ea-9e5d-9b01e8e92c44","name":"Port_7","entity":"PORT","params":{"position":[70000,50000],"portRadius":1000,"height":1100},"xspan":2000,"yspan":2000,"ports":[{"x":0,"y":0,"layer":"FLOW","label":"1"}]},{"id":"8465a6b0-81a6-11ea-9e5d-9b01e8e92c44","name":"Port_8","entity":"PORT","params":{"position":[70000,62000],"portRadius":1000,"height":1100},"xspan":2000,"yspan":2000,"ports":[{"x":0,"y":0,"layer":"FLOW","label":"1"}]},{"id":"88226090-81a6-11ea-9e5d-9b01e8e92c44","name":"Port_9","entity":"PORT","params":{"position":[70000,60000],"portRadius":1000,"height":1100},"xspan":2000,"yspan":2000,"ports":[{"x":0,"y":0,"layer":"FLOW","label":"1"}]},{"id":"8d61e9e0-81a6-11ea-9e5d-9b01e8e92c44","name":"Port_10","entity":"PORT","params":{"position":[70000,58000],"portRadius":1000,"height":1100},"xspan":2000,"yspan":2000,"ports":[{"x":0,"y":0,"layer":"FLOW","label":"1"}]},{"id":"91d12ae0-81a6-11ea-9e5d-9b01e8e92c44","name":"Port_11","entity":"PORT","params":{"position":[70000,55000],"portRadius":1000,"height":1100},"xspan":2000,"yspan":2000,"ports":[{"x":0,"y":0,"layer":"FLOW","label":"1"}]},{"id":"98cd5850-81a6-11ea-9e5d-9b01e8e92c44","name":"Port_12","entity":"PORT","params":{"position":[70000,52000],"portRadius":1000,"height":1100},"xspan":2000,"yspan":2000,"ports":[{"x":0,"y":0,"layer":"FLOW","label":"1"}]},{"id":"e78e9a00-81a9-11ea-9e5d-9b01e8e92c44","name":"Pump_2","entity":"PUMP","params":{"position":[112000,77000],"length":300,"width":2000,"height":250,"spacing":1000,"flowChannelWidth":1000},"xspan":2300,"yspan":2300,"ports":[]},{"id":"f71472b0-81a9-11ea-9e5d-9b01e8e92c44","name":"Pump_3","entity":"PUMP","params":{"position":[104000,45000],"rotation":90,"length":300,"width":2000,"height":250,"spacing":1000,"flowChannelWidth":200},"xspan":2300,"yspan":2000,"ports":[]},{"id":"f8a28cc0-81a9-11ea-9e5d-9b01e8e92c44","name":"Pump_4","entity":"PUMP","params":{"position":[104000,48000],"rotation":90,"length":300,"width":2000,"height":250,"spacing":1000,"flowChannelWidth":200},"xspan":2300,"yspan":2000,"ports":[]},{"id":"fe314780-81a9-11ea-9e5d-9b01e8e92c44","name":"Pump_5","entity":"PUMP","params":{"position":[104000,50000],"rotation":90,"length":300,"width":2000,"height":250,"spacing":1000,"flowChannelWidth":200},"xspan":2300,"yspan":2000,"ports":[]},{"id":"ffe008f1-81a9-11ea-9e5d-9b01e8e92c44","name":"Pump_6","entity":"PUMP","params":{"position":[104000,52000],"rotation":90,"length":300,"width":2000,"height":250,"spacing":1000,"flowChannelWidth":200},"xspan":2300,"yspan":2000,"ports":[]},{"id":"01ca4cc0-81aa-11ea-9e5d-9b01e8e92c44","name":"Pump_7","entity":"PUMP","params":{"position":[104000,55000],"rotation":90,"length":300,"width":2000,"height":250,"spacing":1000,"flowChannelWidth":200},"xspan":2300,"yspan":2000,"ports":[]},{"id":"087cc700-81aa-11ea-9e5d-9b01e8e92c44","name":"Pump_9","entity":"PUMP","params":{"position":[104000,60000],"rotation":90,"length":300,"width":2000,"height":250,"spacing":1000,"flowChannelWidth":200},"xspan":2300,"yspan":2000,"ports":[]},{"id":"09ba2900-81aa-11ea-9e5d-9b01e8e92c44","name":"Pump_10","entity":"PUMP","params":{"position":[104000,62000],"rotation":90,"length":300,"width":2000,"height":250,"spacing":1000,"flowChannelWidth":200},"xspan":2300,"yspan":2000,"ports":[]},{"id":"185a5ca0-81aa-11ea-9e5d-9b01e8e92c44","name":"Pump_11","entity":"PUMP","params":{"position":[104000,58000],"rotation":90,"length":300,"width":2000,"height":250,"spacing":1000,"flowChannelWidth":200},"xspan":2300,"yspan":2000,"ports":[]},{"id":"af05ae70-81aa-11ea-9e5d-9b01e8e92c44","name":"Pump_1","entity":"PUMP","params":{"position":[121000,77000],"length":300,"width":2000,"height":250,"spacing":1000,"flowChannelWidth":200},"xspan":2000,"yspan":2300,"ports":[]},{"id":"cf6cf830-81aa-11ea-9e5d-9b01e8e92c44","name":"Port_1","entity":"PORT","params":{"position":[101000,73000],"portRadius":1000,"height":1100},"xspan":2000,"yspan":2000,"ports":[{"x":0,"y":0,"layer":"FLOW","label":"1"}]},{"id":"d25cc520-81aa-11ea-9e5d-9b01e8e92c44","name":"Port_2","entity":"PORT","params":{"position":[101000,77000],"portRadius":1000,"height":1100},"xspan":2000,"yspan":2000,"ports":[{"x":0,"y":0,"layer":"FLOW","label":"1"}]},{"id":"d9df60a0-81aa-11ea-9e5d-9b01e8e92c44","name":"Port_3","entity":"PORT","params":{"position":[101000,81000],"portRadius":1000,"height":1100},"xspan":2000,"yspan":2000,"ports":[{"x":0,"y":0,"layer":"FLOW","label [... cut ...] ures":{"7fe95c70-819d-11ea-9e5d-9b01e8e92c44":{"id":"7fe95c70-819d-11ea-9e5d-9b01e8e92c44","name":"7fe95c70-819d-11ea-9e5d-9b01e8e92c44","macro":"EDGE","set":"Basic","dxfData":[],"type":"EDGE"}}},{"name":"control","params":{"z_offset":0,"flip":false},"features":{"7fe95c70-819d-11ea-9e5d-9b01e8e92c44":{"id":"7fe95c70-819d-11ea-9e5d-9b01e8e92c44","name":"7fe95c70-819d-11ea-9e5d-9b01e8e92c44","macro":"EDGE","set":"Basic","dxfData":[],"type":"EDGE"}}}],"version":1,"groups":[],"customComponents":{}}`

MurrayLF, 2021-04-23:
I am encountering a similar issue. Regardless of which svg export option I use, I receive essentially blank files. Some information is being recorded; changes I made to the dimensions of the device as a whole were reflected in the svg files and the number of files differed slightly based upon whether the control layer was modified or not. However, no information related to the parts used or image is generated. The JSON, however, appears to be exporting the information properly.

This error was encountered using both the 3Duf website and the HTML from Github. I also updated my chrome version, and this had no effect. The same issue was encountered regardless of which svg export option was used (i.e.; CNC, laser cutting, vector art).

[image]

[image]

Operating System: MacOS Catalina
Browser: Chrome Version 89.0.4389.90, Version 90.0.4430.85

rkrishnasanka, 2021-06-30:
Seems like this is just a problem of not opening the files in inkscape.

closed 2021-06-30 by rkrishnasanka