CIDARLAB/3DuF #192: Fluid as alternative to Mathematical model of Artificial neuron

imvetri, 2020-10-12:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when modern Artificial intelligence uses neurons to train instead of fluids

**Describe the solution you'd like**
Pressure to a fluid is the impulse to a neuron, shape and the path of a fluid's container is a memory. 

**Describe alternatives you've considered**
Use fluid design as an alternative to mathematical model of artificial neurons

**Additional context**
I tried to contact CIDAR after watching this video. https://www.youtube.com/watch?v=aHvfEOlh_b4. I couldn't find any email to send this message to, instead opening an issue to communicate what great potential your lab's work is in the modern world of Artificial intelligence.

As a hobby I had been writing my thoughts out in form of a readme doc in github related to artificial intelligence. As days went by, I found convincing that fluids can replace artificial neurons in machine leaning and my main motivation is to get rid of math involved in modern coding. 

My most recent writing is this https://github.com/imvetri/artificial-intelligence/blob/master/Self.Observing.model.md. Since it doesn't communicate clearly the thoughts I had, I made a video reading through the page with more details to it. Here is the audio recorded with video https://vimeo.com/user101685652.

I'm looking forward to hear from you,
Thanks,
Vetrivel.

rkrishnasanka, 2020-10-28:
Hi @imvetri I'm closing this issue because its not related to 3DuF. However feel free to message me on Twitter `@rkrishnasanka` , I can point you to a couple of papers where microfluidics were used to solve a computational problem. My initial thoughts are :
Approach 1 - You need to formulate whatever optimization problem you setup with machine learning as a fluidic problem. The challenge here would be to translate the entire optimization algorithm into fluidic primitives.
Approach 2 - You want to model the neurons (non-linearity) using the fluid dynamics, so you can look at the shape / form of the final final output as the solution. I believe Photonic Neural Networks might have similar topologies (I can be wrong here). For this approach the challenge would be to convert the digital inputs into a fluidic `signal` and the fluidic `output` back into a digital format.

closed 2020-10-28 by rkrishnasanka

imvetri, 2020-10-28:
Hi @rkrishnasanka. Thanks, the explanations greatly helped.