CIDARLAB/3DuF #193: Sometimes there are no rotation part for a valve in the json file given by 3duf.

zyrrron, 2020-10-31:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

Or add a GIF

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.
[
[Device-3-constant-feeding-1.txt](https://github.com/CIDARLAB/3DuF/files/5469231/Device-3-constant-feeding-1.txt)

](url)

rkrishnasanka, 2021-07-02:
Hey @zyrrron can you describe this issue better ?

zyrrron, 2021-07-03:
It has been a long time and I cannot open this text now. So I also don't know what I really want for this bug. Just forget about it, if I find any other problem I will open a new issue in github.

closed 2021-07-06 by rkrishnasanka