CIDARLAB/3DuF #220: Fixing issues with the component designs and their port locations

rkrishnasanka, 2021-02-06:
@ryu-bu it looks good, I some slight modifications:

1. Not getting the ports for the bottom valve -  I'm not sure why we don't draw channels when we have 4 or 8 outs
[image]
[image]

2. Turns out that I forgot to have some channels to the transposer - green (it'll have control ports - pink)
[image]

3. Fixes to the mux3d component (change the circles to be the component ports - green) (Add red circles for valve - pink)
[image]

ryu-bu, 2021-02-08:
made a PR #221  for this issue.

rkrishnasanka, 2021-02-10:
Looks like everything is cleared now 🥇

closed 2021-02-10 by rkrishnasanka