CIDARLAB/3DuF #229: Create Connection Dialog in Vue.js Framework

rkrishnasanka, 2021-03-17:
This dialog will need to be a variant of property-drawer and have the capability for users to change the connection setting during drawing

vyiu, 2021-06-10:
I'm working on this

rkrishnasanka, 2021-06-23:
To activate Connection tool check - Line-186 `componentToolBar`

Check `connectionTool` to get the data about the connection

rkrishnasanka, 2021-07-01:
This line should activate it`Registry.viewManager.activateTool("Connection", "Connection");`

rkrishnasanka, 2021-07-09:
@vyiu I fixed the issues with the mouseclicks not registering. Changes that need to be made:

- [ ] You should update the dialog so that it opens at a fixed position on the canvas (top) as shown in the mockup. Right now it's opening wherever the button is.

- [ ] You should fix the width of the dialog window to be smaller than the canvas and potentially reduce the size of the profile panel and the sink+source panel.