CIDARLAB/3DuF #238: Website Freezing/Crashing when Switching Between Control and Flow Layers

JoshuaSingh33, 2021-04-18:
**Describe the bug**
A clear and concise description of what the bug is.
When I try to switch between the control and flow layer (by clicking their respective buttons), my 3duf crashes.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'select layer'
2. Click on 'control' or 'flow' (depends on what current layer is, want to switch to other, non-current layer).
3. See error

Or add a GIF

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Windows
 - Browser [e.g. chrome, safari] Chrome
 - Version [e.g. 22] 10

**Additional context**
Add any other context about the problem here.

JoshuaSingh33, 2021-04-21:
I wasn't able to attach my .json file as it said that file type is not supported (I will have to email it to you as it is too big for piazza attachment).

msuazox, 2021-04-22:
I'm also having problems with the website being unresponsive. When I try to use the connection tools, nothing shows. At first I thought I was using it wrong, but in another tab I tried 3duf and used the connection tab first and it works. Then after a while it stops working again. 

Desktop 
- OS 
- Chrome 
- Big Sur 11.2.3

Similarly when I use the command v function, the website undos connections instead of pasting. 

https://user-images.githubusercontent.com/61927568/115661815-098b6f00-a30c-11eb-9fd7-bda34eeb987c.mov

msuazox, 2021-04-23:
[image]

rkrishnasanka, 2021-06-18:
Any luck uploading the Zip file (containing the json) for this ?

rkrishnasanka, 2021-08-19:
This is being fixed in v2

closed 2021-08-19 by rkrishnasanka