CIDARLAB/3DuF #249: SVG File Download

sebriquet, 2021-04-23:
**Describe the bug**
When trying to download the SVG files from 3Duf, the system downloads blank files  they show blank in chrome as well as with SVG viewer software.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 3Duf and complete your design
2. Click on SVG download
3. Open downloaded files (eg use Gapplin if using MAC)
4. See blank file

**Expected behavior**
The SVG file should be visible either in the URL or with a SVG viewer.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: iOS  (macOS Catalina)
 - Browser : Chrome 
 - Version: Version 88.0.4324.150 (Official Build) (x86_64)

**Additional context**

rkrishnasanka, 2021-04-26:
@sebriquet could you try opening the SVG with inkscape (https://inkscape.org/) ? Upload the SVGs so that I can see if its just an issue with the browser not opening them properly

sebriquet, 2021-04-26:
Hi Krishna,

Attached are the SVG files, I could not open them with inkspace either.

rkrishnasanka, 2021-04-26:
@sebriquet You'll need to upload the svg on the github, I don't think it detects the attachments

sebriquet, 2021-04-27:
GitHub is not letting me upload svg files: 
[image]

rkrishnasanka, 2021-04-27:
Create a zip ?

rkrishnasanka, 2021-06-17:
@sebriquet could you make a zip file of this design?

rkrishnasanka, 2021-07-02:
Hey @sebriquet it seems like most of the svg issues are because you aren't opening the svg in Inkscape. I'm gonna close this issue since I don't have any way to check and see if the svg is working or not.

closed 2021-07-02 by rkrishnasanka