CIDARLAB/3DuF #263: [Story] Create Common Eventbus communication interface

rkrishnasanka, 2021-06-04:
### Description

Since we are moving to Vue, we will be handling all the event communication between the core Vue UI components and the core components via the eventemitter3 (https://github.com/primus/eventemitter3) library.

Create EventBus System in `/src/app/view` folder:
1. Create event receiver stubs for all the + Vue UI user-driven events 
2. Create event dispatcher stubs for the canvas interactions
3. Implement dispatcher calls from the hotkey, double-click UI etc.

Acceptance Criteria:
- [ ] Ensure that all the events that arise from canvas interactions are captured here
- [ ] Ensure that all the events that arise from UI interactions are captured

MurrayLF, 2021-06-09:
+

MurrayLF, 2021-06-14:
EventBuses defined in events.ts for handling of parameter menus and grid resolution/size change.
1. UPDATE_GRID_SIZE emitted in ResolutionToolbar.vue, currently no listeners (.vue file not yet written)
2. SHOW_COMPONENT_PARAMS, SHOW_CONNECTION_PARAMS, and EDIT_CONNECTION defined but have neither emits nor listeners. 
No .vue files currently exist for ui/*, likely need to be made before instances of these events are useable.

rkrishnasanka, 2021-07-22:
This one is done, no PR was linked to this

closed 2021-07-22 by rkrishnasanka