CIDARLAB/3DuF #284: [Story] Enable Detailed Error handling + report generation for imported JSON

rkrishnasanka, 2021-06-17:
### Description

On uploading the JSON, the user should receive a detailed error report if any of the JSON is malformed or broken.

- [ ] Should print out a detailed list of what parameters are missing per component/connection, what parameters need to be inserted for each component/connection
- [ ] Should incorporate json cleanup utils already present in `utils`
- [ ] Should accommodate for design rule parameters such as component spacing
- [ ] Should validate against parchmint schema and print out all the warnings

**Additional Description:** 

(Attach mockups and more information here)

Acceptance Criteria:
- [ ] Unit Test
- [ ] Should be written in typescript

rkrishnasanka, 2021-06-17:
@jkamoona I'll need you to make a mockup for how this is supposed to look like

@murrayLF This will be the extension of the typescript port you are doing right now

MurrayLF, 2021-06-18:
-