CIDARLAB/3DuF #371: **Ability to create curved channels**

rkrishnasanka, 2021-08-11:
Background
Currently, 3DuF only has the ability to create channels/connections that are straight or have sharp edges. 

Goal
Add the drawing code in paperjs that can generate Paths using either the RobustPath or the FlexPath classes seen in the python library `gdstk` 
(https://heitzmann.github.io/gdstk/gettingstarted.html#paths). This will add new tool buttons called "Robust Path Connection" and "Flex Tool Connection". 

**Note: These will be integrated into a single connection tool at a later point**

Difficulty Level 2
requires the ability to translate c++ implementation into structures created by paper.js

Skills
- Ability to understand C++ implementations and translate them into javascript
- Familiarity with how 3DuF user tools are created
- Familiarity with paper.js library

Potential Mentors
TBD

Contact
[Krishna](mailto:rkrishnasanka@gmail.com)

nmz787, 2021-08-25:
I reviewed the links above and compared the APIs from these examples:
http://paperjs.org/tutorials/paths/working-with-path-items/
http://paperjs.org/tutorials/paths/using-color-and-style/
https://heitzmann.github.io/gdstk/gettingstarted.html#paths

and noticed that the basic common requirements for a curved path in either library are:
* list of points
* width

While on call with Michael, I suggested that he start with a simple dictionary with a structure like this:
```
{
  'points': [(0, 0), (3, 0), (3, 2), (5, 3), (3, 4), (0, 4)],
  'width': 5
}
```

and then write code in either language that consumes this common data source. Using a dictionary will allow for serialization to save designs to disk, for reloading later in either Paper.JS or gdstk (for export to GDSII for fab consumption). As more design requirements are discovered, it should allow easy additions.

During serialization, I *think* all curved-path objects would need packaged together, so it could be properly parsed and appropriate language-specific API calls made. Something like:
```
{
  'flexpaths': {
    'path_name_or_id_1':
      {
        'points': [(0, 0), (3, 0), (3, 2), (5, 3), (3, 4), (0, 4)],
        'width': 5
      },
    'path_name_or_id_2':
      {
        'points': [(30, 75), (30, 25), (80, 25),(80, 75)],
        'width': 3
      }
   }
}
```

Hopefully that's on the right track!

rkrishnasanka, 2021-08-25:
@nmz787 Followup question - Is there any way to track the bend radius in this ?