CIDARLAB/3DuF #470: v2.3duf

zyrrron, 2023-03-08:
[image]

cannot use connection after click on it.

rkrishnasanka, 2023-03-08:
The current version of `fixes/valveinsertion` partially addresses this issue of the renders breaking. We needed to add a new system for rendering channel breaks using the geometries/features with the subtract macros. The temporary fix prevents the errors that make the execution buggy. I'll merge the PR with the fix so all the features get added and tracked correctly.

rkrishnasanka, 2023-03-09:
@zyrrron I cleaned up some of the code in the area since I couldn't actually reproduce the error. Seems like I couldn't figure out the exact actions you did before it failed.

zyrrron, 2023-03-09:
renderLayer.ts:129 Uncaught Error: Layer does not contain a feature with the specified ID!
    at e.value (renderLayer.ts:129:55)
    at e.value (renderLayer.ts:139:14)
    at n.value (connectionTool.ts:196:69)
    at e.<anonymous> (connectionTool.ts:83:21)
    at e.value (simpleQueue.ts:26:18)
    at i.move (connectionTool.ts:143:33)
    at HTMLCanvasElement.<anonymous> (mouseAndKeyboardHandler.ts:309:17)